GEIGEIGEIST/KLOR #5: How to set side

rklasen, 2022-07-24:
Hi, thanks for this fantastic project!

I'm currently building one of you KLOR Keyboards, this is my first DIY keyboard. Also my first time using QMK. Everything seems to be working for now, except for one thing.

How do I set the sides? It seems the side with the USB connection is always considered "left", but I'd like to connect the USB cable to the right side.

thanks again, and best wishes!

GEIGEIGEIST, 2022-07-24:
Hi R Klasen,
You should be able to add
`#define MASTER_RIGHT`
to the config.h file of your keymap. (for example klor\keymaps\default\config.h)
Please tell me if this works and I would be happy to see some pictures of it, when you finished it.

rklasen, 2022-07-24:
Hi, thank you for the quick answer! Yep, that worked!

I'll post some photos in this issue when I'm done, but my version won't be nearly as pretty as yours :sweat_smile: 

I've just looked though the [qmk docs](https://github.com/qmk/qmk_firmware/blob/master/docs/feature_split_keyboard.md), and it seems one can set the headedness via a pull up o pull down. I'm not familiar enough with your circuit, so I don't know if any µC pin is still left available, but if it is, it should be possible to use `#define SPLIT_HAND_PIN B7` to set the left side during soldering. Either via solder bridge or µC ground. Maybe for v1.4 :smile: 

So, thanks again, I really love this project!

GEIGEIGEIST, 2022-07-24:
Great it worked!

Hehe it's not a battle, I'm just really curious to see some people actually build it.

Yea, I know this one, but thank you for mentioning it. Unfortunately I've already used all the available pins for this version. I have some ideas for a next one, which would even use the pins available on MCUs with pins on the bottom.

closed 2022-08-22 by GEIGEIGEIST