GEIGEIGEIST/KLOR #6: the stl files abnornal

hackqiang, 2022-08-01:
when I try to print the case via my printer, I found the stl file is abnormal while slicing(both checked in superslicer and Cura).
ex: KLOR_konrad_case_ble_L.stl in case\3DP\konrad\bluetooth:
[image]

hackqiang, 2022-08-01:
I can use the fusion 360 to fix: open stl, fix model and export stl.

GEIGEIGEIST, 2022-08-01:
Thank you for pointing this out. May I ask which fix you applied? (Stitch and Remove/Close Holes/Wrap/Rebuild)
Unfortunately the case was made in Blender, since it was easier to create all the variations for me using Geometry Nodes.

hackqiang, 2022-08-07:
I'm sorry for reply so late, I use "stitch and remove" to fix the STLs:
[image]

hackqiang, 2022-08-07:
and for case of bluetooth verison, it seems no space for battery

GEIGEIGEIST, 2022-08-22:
I finally fixed all the STLs. 

But yea, I still didn't found a good place for the battery other than under the MCU. Originally it was meant to be a QMK only board, but some people asked me for JST and power-switches.

closed 2022-08-23 by GEIGEIGEIST