GEIGEIGEIST/KLOR #7: Keys not working with minimum viable example

sommerper, 2022-08-19:
Hi!

Thank you for a great project.

I'm having some trouble getting a minimum viable version to work.
[image]

As seen in the included image I first tried soldering all the components but that didn't work. No keys, LEDs, knob, nor the LCD worked.

Then I tried making a version where I would solder all the diodes but mount just one key - but that also didn't type anything when I pressed the key.

I've tested all the diodes with a multimeter and they all seem to be mounted correctly.

I've tested with several Pro Micros fresh out of the bag.

The first time I flash I can use USB-C with this command:

```shell
avrdude -v -patmega32u4 -cavr109 -P/dev/ttyACM0 -b57600 -Uflash:w:"/home/xxx/qmk/qmk_firmware/klor_default.hex":i
```

Then subsequently the Pro Micro will not show up on 'ttyACM0' anymore (even when double tapping reset) and I have to flash using an 'Arduino as ISP'. Like so:

```shell
/home/xxx/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/bin/avrdude "-C/home/xxx/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf" -v -patmega32u4 -cavr109 -P/dev/ttyACM0 -b19200 -cstk500v1 -Ulock:w:0x2F:m -Uflash:w:"/home/xxx/qmk/qmk_firmware/klor_default.hex":i
```

An odity - but in any case - it works and the Pro Micro shows up as `Bus 001 Device 125: ID 3a3c:0001 GEIST KLOR`.

My `keymaps/default/rules.mk` looks like this:

```shell
OLED_ENABLE = no
OLED_DRIVER = SSD1306
ENCODER_ENABLE = yes
EXTRAKEY_ENABLE = yes
AUDIO_ENABLE = no
MUSIC_ENABLE = no
MOUSEKEY_ENABLE = yes

DYNAMIC_MACRO_ENABLE = yes
COMBO_ENABLE = yes
KEY_OVERRIDE_ENABLE = yes

#HAPTIC FEEBACK
HAPTIC_ENABLE = no

#PER KEY RGB
RGB_MATRIX_ENABLE = no
```

I'm kinda at a loss here but I suspect it's a QMK configuration problem since I traced the paths with a multimeter and there where no interruptions from switches to the Pro Micro.

I'm not a complete noob when it comes to soldering keyboards as I've done a few Ferris' and Plancks but I won't leave out the possibility that I've done something wrong while soldering.

Any suggestions as to what could be wrong are greatly appreciated!

sommerper, 2022-08-21:
Out of curiosity I tried flashing a Contra.hex (Planck layout) and I could get some of the keys to work. So I know that at least the soldering works in most places. After flashing the Klor firmware back on the keys where all dead again. 

Which, again, makes me think this is a QMK configuration issue.

GEIGEIGEIST, 2022-08-21:
Sorry for answering so late, I'm a little bit busy right now. But if you like to I could send you a working hex file, so you could see if it's a hardware or software issue.
By the way depending on your case the pro micro sockets seem a bit high

sommerper, 2022-08-21:
Thanks for replying and no worries. I'm not in a hurry.
The sockets are just temporary because I ran out of pins.
[image]

But, yes! If you can find the time to upload a working .hex it would be appreciated!

GEIGEIGEIST, 2022-08-21:
You can find it [here](https://github.com/GEIGEIGEIST/qmk-config-klor)

sommerper, 2022-08-21:
Unfortunately that also didn't work. But thank you for taking the time!

I've double checked all the traces to the pins of the pro micro and all seems fine.

This might be a stupid question but can I check the left keyboard alone without having the right one connected? So far I've only made the left side.

GEIGEIGEIST, 2022-08-21:
Oh than it seems to be an hardware issue. 
Unfortunately hard to tell where the problem could be. Do you own a multimeter?

And yes you can check both halves separatly.

sommerper, 2022-08-21:
I do indeed have a multimeter. That's how I tested the traces and diodes earlier.
Since the Vial Contra made some of the keys respond I tried fiddling with the code a little - and it turns out that if I switch the matrix configuration around like this
```
// wiring of each half
// #define MATRIX_ROW_PINS { C6, D7, E6, B4 }
#define MATRIX_ROW_PINS { F6, F7, B1, B3 }
// #define MATRIX_COL_PINS { F6, F7, B1, B3, B2, B6 }
#define MATRIX_COL_PINS { C6, D7, E6, B4, B2, B6 }
```
a lot of keys start working but with a weird layout.
I starting to wonder if I soldered the Pro Micro in the wrong holes.
But judging from the [photo](https://github.com/GEIGEIGEIST/KLOR/blob/main/docs/images/buildguide/breakoff.jpg) I did do it correctly. Basically following the white outlines.
[image]
I'll see if I can figure out some more.

sommerper, 2022-08-21:
Indeed, following [this guide](https://deskthority.net/wiki/Arduino_Pro_Micro#Pinout) I basically mirrored the default config and got all of the keys working:

```
// wiring of each half
// #define MATRIX_ROW_PINS { C6, D7, E6, B4 }
#define MATRIX_ROW_PINS { F7, B1, B3, B2 }
// #define MATRIX_COL_PINS { F6, F7, B1, B3, B2, B6 }
#define MATRIX_COL_PINS { D4, C6, D7, E6, B4, B5 }
```
I still haven't figured out the settings for the encoder and OLED display.

But I guess it boils down to having soldered the Pro Micro either on the wrong side or in the wrong holes?

GEIGEIGEIST, 2022-08-21:
Sorry I didn't noticed before, but you got it mounted the wrong way. The components needs to face the PCB.

sommerper, 2022-08-22:
Damn, you're absolutely correct. It's even clearly mentioned in the build guide. I knew it was probably something stupid on my part. Still haven't tested OLED and LEDs but all keys and encoders work now.

Sorry for wasting your time :-/

closed 2022-08-22 by sommerper

GEIGEIGEIST, 2022-08-22:
Hehe don't worry, stuff like this happens sometimes and I didn't noticed at first either. Glad it's working now.

sommerper, 2022-08-23:
It's looking and working really good!
[image]

Would you be willing to open the discussions forum? I have a question that is not an issue. Or should I just ask here?

GEIGEIGEIST, 2022-08-23:
Oh just ask here. And by the way it looks really nice!

sommerper, 2022-08-23:
I just wanted to know what the timeline for Vial compatibility looks like. For me, flashing with QMK every time is very time consuming process because I can't get the chip to go into bootloader mode and I have to use an AVR flasher. And with every new keyboard I spend a good amount of time optimizing a layout.

Maybe I can try to make the Sofle template in the repository work somehow.

GEIGEIGEIST, 2022-08-23:
While someone in my main discord channel is doing a VIAL workshop today it probably will take some time to do it right. But you can use this hex in the meantime (which was done by someone who already knows how to handle VIAL
[klor_vial.zip](https://github.com/GEIGEIGEIST/KLOR/files/9401809/klor_vial.zip)
)

sommerper, 2022-08-23:
Thanks! That already works pretty well! Let me know if you need any testing done before releasing it.

GEIGEIGEIST, 2022-08-23:
I maybe will, thank you