GEIGEIGEIST/KLOR #21: Clarification on Switchplates and Concerns about 3D Printing Warning

sedghi, 2023-10-25:
I read 

> You need a switchplate and the acrylic parts, no matter what case you choose.

But I believe you mean two switchplates right? 

When I upload the stl file to jlcpb I see the following warning for the 3D printing, any combination of the 3D technology and material I use still shows it. Any idea if this is serious warning?

`The minimum build size for LEDO 6060 Resin is 1.00x1.00x0.20cm, please change the material.`

[image]

sedghi, 2023-10-25:
Thanks for your reply. 
Seems like the FR-4 does not give that error for 3D print, so I will probably do that. 

I requested a quote from a local shop around me for the Archlyc. Thanks !

sedghi, 2023-10-25:
@CipherJay Which layout and which version (BLE vs wired) you built out of curiosity

LucaEic, 2023-11-15:
Hi everyone,

I'm also looking into building the Polydactyl. 

Regarding 2) It looks like the length x width x heigth dimension are in the wrong order. It say "minimum is 1.00 x 1.00 x 0.2 cm and the given dimension in the sketch are 15.03 x 0.15 x 10.48 cm. Howver, I'm not fond enough to change this issue thats why i looked in the FR-4 switchplat.

This being said, I have a question to the FR-4 switchplate: when uploading the gerber files the preview looks like this: 

[image]

Are the cut outs for the keys present? Or is this just a display issue that it appears to be a solid plate? Furthermore, the 3D printed parts shall be 1.5mm however the FR-4 can only be 1.2 or 1.6mm. Does this have any effect? I would like to use silent tactile switches (like Boba U4t). I have read that for the tiny choc switches 1.2mm should be used. 

Thanks and chers
Luca

Vinna152, 2023-11-20:
I'd advise looking at the project files in KiCad to confirm. The exported gerbers should be fine since they're based on those files.

These Gerber previews often aren't 100% representative of the actual file structure. 
You can also confirm the exact sizes in KiCad and even if it's just an estimate an engineer on their side will correct it for you.