GEIGEIGEIST/KLOR #27: Issues with JLCPCB?

jesseleite, 2024-05-02:
Anyone have trouble getting the KLOR printed with JLCPCB?

[image]

Seems odd that they would suggest short circuits in the KLOR gerbers? Am I doing something wrong maybe? All settings are default, except that I selected the color black, and 'yes' to remove order number.

jesseleite, 2024-05-03:
Oh they responded to my customer support request. They're saying:

>"The distance from copper to slot should be at least 0.3 mm"

@GEIGEIGEIST By chance have you had any trouble getting these printed at JLCPCB? They attached these pics to the support convo :'(

[image]
[image]

Lefuneste83, 2024-07-05:
> Oh they responded to my customer support request. They're saying:
> 
> > "The distance from copper to slot should be at least 0.3 mm"
> 
> @GEIGEIGEIST By chance have you had any trouble getting these printed at JLCPCB? They attached these pics to the support convo :'(
> 

Hi there. I have bumped into the very same issue today, or at least I noticed something was totally wrong before placing an order with JLCPCB. The problem you encountered does not come from "shorted tracks" but rather from the routing around the LED cutouts. When I imported the gerbers I noticed many LED cutouts were missing, and reading your post I made the connection...

As you may have noticed this PCB design is reversible and thus both layouts for Left and Right boards have to share the same vertical space.
On first analysis this approach may sound clever, but considering the added routing complexity and the rather low cost of manufacturing a set of PCB, it may be better to implement a dedicated PCB for Left and Right boards rather than to attempt cramming everything into one board.
Anyway, I have reviewed the design and indeed many traces are not clearing the LED cutouts enough to be able to manufacture them. 
If I may give my opinion,I find the routing rather weird, and maybe there was some auto routing involved at some point. From my experience in designing PCB and being as OCD as possible about trace clearance and proper routing, I think there may be room for improvement with  this PCB design.  I wish to emphasis that this does NOT relate to the quality of the overall project. It just makes it a little bit more difficult to tame. I haven't been as far as to check DRC yet using JLCPCB set of rules, so there may be other pending issues.
I will try to fork the repo and update the most daring issues. In the meantime let me know if you need updated gerbers that will clear JLCPCB review process.

Korkd, 2024-07-08:
> I will try to fork the repo and update the most daring issues. In the meantime let me know if you need updated gerbers that will clear JLCPCB review process.

Any chance you've gotten around to posting these?

Korkd, 2024-07-09:
[KLOR_pcb.zip](https://github.com/user-attachments/files/16140852/KLOR_pcb.zip)

Took a shot at cleaning up the PCB. If anyone more experienced/knowledgeable has a chance to look at it and let me know if there are any issues that'd be great. Haven't done any PCB design before so just a buyer beware if you stumble upon this.

Lefuneste83, 2024-07-09:
Here is my take on the rerouted PCB. If you wish to rework the PCB from the native Kicad files, beware that the various silkscreen can end up on top of the copper pads if you forget to tick the proper box when exporting gerber files.
I have ordered the following file and it was validated by JLCPCB although the DRC (with proper DRU file from JLCPCB) fails with hundreds of warnings and errors due to the nature of the design (cutouts, clearances...). I hope the PCB still is OK but I don't see why it wouldn't. I haven't built it yet so be careful. The native modified files should be on my fork.

https://github.com/Lefuneste83/KLOR/blob/main/PCB/klor1_3/gerbers/gerbers.zip

jesseleite, 2024-08-19:
@Lefuneste83 Curious if you ended up building yours off those gerbers you linked, and how it worked out? 🙏

jesseleite, 2024-08-19:
@Lefuneste83 Curious if you ended up building yours off those gerbers you linked, and how it worked out?

>I have ordered the following file and it was validated by JLCPCB although the DRC (with proper DRU file from JLCPCB) fails with hundreds of warnings and errors due to the nature of the design (cutouts, clearances...)

Oh... So even with your rerouting changes, JLC still gave you warnings and errors around cutouts/clearances like in my original post's screenshots?

Lefuneste83, 2024-08-19:
> @Lefuneste83 Curious if you ended up building yours off those gerbers you linked, and how it worked out?
>
>
> Oh... So even with your rerouting changes, JLC still gave you warnings and errors around [...]

Yes original design 1.3 will give you several kind of DRC errors mainly due to clearances on switch holes and in the way markings are made with mask/copper layers. Design 1.4 being rerouted will not have the same errors. With the included JLCPCB dru file you should only get a minor error due to termination of the LED strip which has a pair of unconnected nodes which is to be expected. Both updated 1.3 and 1.4 on my fork will be built to functional boards by JLCPCB. I have ordered and built both designs. I suggest you get the 1.4 design using SK6812 Mini-e LED package which are way easier and cleaner to build.

jesseleite, 2024-08-19:
I'm not even using LEDs on this build, but if 1.4 is less JLC warnings/errors, that sounds great! I see your commit on the [1.4 gerbers.zip](https://github.com/Lefuneste83/KLOR/blob/main/PCB/klor1_4/gerbers/gerbers.zip) is only from last week. Do you plan on getting them printed and into a build soon? Curious how it goes, and if any issues ❤️

Lefuneste83, 2024-08-19:
I have built my custom 1.3 version (the one found on my fork) and the 1.4 version (the version just before the very last commit) and I am currently using it as my keyboard. Both work perfectly. As explained in the changelog I discovered a small interference of about 0.45 mm between the edge of the stock haptic module PCB (which I currently do not have because I find it too expensive) and the top switch socket. The latest 1.4 commit corrects this by moving the haptic module position and cutout by 0.6 mm up in order to clear the interference. This interference comes from the reversed switch position to allow for south facing of the LED. I have also included a DIY version of the haptic module which I will be validating soon. Just waiting for parts and PCB. All these modifications only concern the haptic module itself. If you don't intend to use it then even the previous commit will be fine. It will only cause a small issue if using the stock Pimoroni module but not the DIY one I designed which will work even on previous commit as it has been shorten by 0.8 mm. Even in the worst case scenario of not using the very latest commit of the 1.4 and the Pimoroni module, the interference is so small you can just file the PCB of the haptic module by a tiny bit to clear it. It may even not be an interference at all depending of various mechanical tolerances. This is just to make the most perfect PCB possible. In case you put LED anyways the new packages are super cheap and really easy to solder. Way easier than WS2812 3535. Even if not using them for "light shows" they are nice to have as they can be programmed to indicate active layer at least if using QMK. If you will never solder any LED then any version 1.3 or 1.4 will be similar for you. The 1.4 will definitely have less DRC errors as I have made sure to correct them and this was not an easy task.