GEIGEIGEIST/KLOR #28: Compile fails.

pokeygithub, 2024-06-27:
☒ Not including data from file: keyboards/klor/info.json
☒       layouts.LAYOUT_yubitsume.layout.20.x: -0.5 is less than the minimum of 0
☒ klor/2040: RGB_DI_PIN in config.h is no longer a valid option and should be replaced with WS2812_DI_PIN or APA102_DI_PIN
☒ Invalid API data: klor/2040: haptic.driver: 'DRV2605L' is not one of ['drv2605l', 'solenoid']

 *                                                                                                     [ERRORS]

builddefs/converters.mk:22: *** Converting to 'promicro_rp2040' not possible!.  Stop.

Manamonk, 2024-07-09:
I've got the same errors did you find the answer?

pokeygithub, 2024-07-10:
What I had to do was abandon the 2040 (folder) command and do qmk make -kb klor -km default -e CONVERT_TO=promicro.
You might need edit the config.h in that folder find and replace RGB_DI_PIN with WS2812_DI_PIN and DRV2607L with drv26505l  it hates all caps for some reason!
You will need to edit the kyboard layout if you are trying to use Colmak layout and if you edit the adjust layer command you can run the make command above with a simple key combination to make and then to reset for flashing.
Please let me know if that works for you!

pokeygithub, 2024-07-11:
[klor2.zip](https://github.com/user-attachments/files/16170307/klor2.zip)

If you extract it as klor2 it won't screw up any of your original files.
Maybe this will help you.  If you use the "pokey" it will get colmak working for you as well.

Manamonk, 2024-07-18:
Hi pokey - I tried running the pokey keymap and got the following :(

[dm@hhhhh QMK_files]$ qmk compile -kb klor -km pokey
Ψ Compiling keymap with make -r -R -f builddefs/build_keyboard.mk -s KEYBOARD=klor KEYMAP=pokey KEYBOARD_FILESAFE=klor TARGET=klor_pokey INTERMEDIATE_OUTPUT=.build/obj_klor_pokey VERBOSE=false COLOR=true SILENT=false QMK_BIN="qmk"

☒ Not including data from file: keyboards/klor/info.json
☒       layouts.LAYOUT_yubitsume.layout.20.x: -0.5 is less than the minimum of 0
⚠ klor: Build marker "keyboard.json" not found.
☒ klor: No LAYOUTs defined! Need at least one layout defined in info.json.
☒ klor: LAYOUT_polydactyl: Layout macro should not be defined within ".h" files.
☒ klor: LAYOUT_konrad: Layout macro should not be defined within ".h" files.
☒ klor: LAYOUT_yubitsume: Layout macro should not be defined within ".h" files.
☒ klor: LAYOUT_saegewerk: Layout macro should not be defined within ".h" files.
Generating: .build/obj_klor_pokey/src/info_deps.d                                                   [OK]
☒ Not including data from file: keyboards/klor/info.json
☒       layouts.LAYOUT_yubitsume.layout.20.x: -0.5 is less than the minimum of 0
⚠ klor: Build marker "keyboard.json" not found.
☒ klor: No LAYOUTs defined! Need at least one layout defined in info.json.
☒ klor: LAYOUT_polydactyl: Layout macro should not be defined within ".h" files.
☒ klor: LAYOUT_konrad: Layout macro should not be defined within ".h" files.
☒ klor: LAYOUT_yubitsume: Layout macro should not be defined within ".h" files.
☒ klor: LAYOUT_saegewerk: Layout macro should not be defined within ".h" files.
avr-gcc.exe (GCC) 12.2.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Generating: .build/obj_klor_pokey/src/info_config.h                                                 [WARNINGS]
 |
 | ☒ Not including data from file: keyboards/klor/info.json
 | ☒    layouts.LAYOUT_polydactyl.layout.20.x: -0.5 is less than the minimum of 0
 | ⚠ klor: Build marker "keyboard.json" not found.
 | ☒ klor: No LAYOUTs defined! Need at least one layout defined in info.json.
 | ☒ klor: LAYOUT_polydactyl: Layout macro should not be defined within ".h" files.
 | ☒ klor: LAYOUT_konrad: Layout macro should not be defined within ".h" files.
 | ☒ klor: LAYOUT_yubitsume: Layout macro should not be defined within ".h" files.
 | ☒ klor: LAYOUT_saegewerk: Layout macro should not be defined within ".h" files.
 |
Generating: .build/obj_klor_pokey/src/default_keyboard.c                                            [WARNINGS]
 |
 | ☒ Not including data from file: keyboards/klor/info.json
 | ☒    layouts.LAYOUT_saegewerk.layout.20.x: -0.5 is less than the minimum of 0
 | ⚠ klor: Build marker "keyboard.json" not found.
 | ☒ klor: No LAYOUTs defined! Need at least one layout defined in info.json.
 | ☒ klor: LAYOUT_polydactyl: Layout macro should not be defined within ".h" files.
 | ☒ klor: LAYOUT_konrad: Layout macro should not be defined within ".h" files.
 | ☒ klor: LAYOUT_yubitsume: Layout macro should not be defined within ".h" files.
 | ☒ klor: LAYOUT_saegewerk: Layout macro should not be defined within ".h" files.
 |
Generating: .build/obj_klor_pokey/src/default_keyboard.h                                            [WARNINGS]
 |
 | ☒ Not including data from file: keyboards/klor/info.json
 | ☒    layouts.LAYOUT_saegewerk.layout.20.x: -0.5 is less than the minimum of 0
 | ⚠ klor: Build marker "keyboard.json" not found.
 | ☒ klor: No LAYOUTs defined! Need at least one layout defined in info.json.
 | ☒ klor: LAYOUT_polydactyl: Layout macro should not be defined within ".h" files.
 | ☒ klor: LAYOUT_konrad: Layout macro should not be defined within ".h" files.
 | ☒ klor: LAYOUT_yubitsume: Layout macro should not be defined within ".h" files.
 | ☒ klor: LAYOUT_saegewerk: Layout macro should not be defined within ".h" files.
 |
Compiling: keyboards/klor/klor.c                                                                    [OK]
Compiling: .build/obj_klor_pokey/src/default_keyboard.c                                             [OK]
Compiling: quantum/keymap_introspection.c                                                           [OK]
Compiling: quantum/quantum.c                                                                        [OK]
Compiling: quantum/bitwise.c                                                                        [OK]
Compiling: quantum/led.c                                                                            [OK]
Compiling: quantum/action.c                                                                         [OK]
Compiling: quantum/action_layer.c                                                                   [OK]
Compiling: quantum/action_tapping.c                                                                 [OK]
Compiling: quantum/action_util.c                                                                    [OK]
[... 40 lines cut ...]
Compiling: tmk_core/protocol/usb_descriptor.c                                                      tmk_core/protocol/usb_descriptor.c:449:35: error: 'VENDOR_ID' undeclared here (not in a function)
  449 |     .VendorID                   = VENDOR_ID,
      |                                   ^~~~~~~~~
tmk_core/protocol/usb_descriptor.c:450:35: error: 'PRODUCT_ID' undeclared here (not in a function); did you mean 'PRODUCT'?
[... cut ...]
      |                             ^
tmk_core/protocol/usb_descriptor_common.h:20:25: note: in definition of macro 'USBCONCAT'
   20 | #define USBCONCAT(a, b) a##b
      |                         ^
tmk_core/protocol/usb_descriptor.c:1050:42: note: in expansion of macro 'USBSTR'
 1050 |         .Size                   = sizeof(USBSTR(MANUFACTURER)),
      |                                          ^~~~~~
 [ERRORS]
 |
 |
 |
make: *** [builddefs/common_rules.mk:373: .build/obj_klor_pokey/usb_descriptor.o] Error 1

pokeygithub, 2024-07-21:
Hmmm.  Maybe it I forgot that I modified the json? 
[info.json](https://github.com/user-attachments/files/16322984/info.json)

Are you running everything as an administrator?  Pretty much have to do the entire process from installing QMK to running the keyboard make commands.