GEIGEIGEIST/KLOR #30: Anyone have working files for RP2040?

Manamonk, 2024-07-19:
Does anyone have a working set of files GitHub or download for the latest version of QMK (UF2 for RP2040) preferably with vial or via enabled. Been trying to get the current ones on the git hub working for week but cant seem to get it to work. The default pre compiled works but really need to edit the keys so I can get the shift key on the left hand keyboard so I can use it with games.

Lefuneste83, 2024-08-25:
Hi there. I have just managed to adapt config files to be able to compile QMK firmware on the latest code base for RP2040. Check out [my fork](https://github.com/Lefuneste83/KLOR) for config or precompiled files. It is not perfect but it should compile fine.I still haven't been able to have VIA or VIAL working though... There is still some work to do so...

Lefuneste83, 2024-09-06:
FYI VIAL is now working on RP2040