GEIGEIGEIST/KLOR #34: MCU sockets docu

hoijui, 2025-01-02:
The BoM mentions MCU sockets.
As I am a newb, I was a bit lost with that. After searching quite some, I figured that one possible solution is to look for "DIP 24 wide socket round hole", which should fit the ProMicro, when using diode legs soldered onto the controller board.

I would like to document this in the build-guide, so other newbs don't have to research it.
What do you think?

## Details

important info:

- A Pro micro has 2 rows of 12 pins each -> DIP 24.
- The distance between the legs is 0.1" (2.54mm), which is the standard.
- The distance between the two rows of pins, measured from the center of a pin to the center of a pin on the other row) is 0.6" (15.24mm).
- (redundant but nice to know/check:) distance from the first to the last (12th) pin is 11 * 0.1" = 1.1" (27,94mm)

This is met by "DIP 24 Wide", and the "round hole" sockets work well with the diode legs.

One can find this info sometimes on the site where one can buy the sockets, or looking at [datasheets](https://www.farnell.com/datasheets/217369.pdf) (here specifically the 800 series, page 16 of the PDF, aka page A15).

Lefuneste83, 2025-04-11:
You are right to mention sockets choice as this is definitely the most difficult part of this build. I ended up using low profile round pin sockets but these can be difficult to source. The issue comes mostely from the 3d case design which does not provide enough clearance for a normal Dupont style socket heighth. Make sure you test clearance before soldering if you wish to use socket mount. In most cases I would recommand not using sockets but first builders may end up having problems unsoldering the MCU if something goes wrong so it is a bit of a pain to give a definitive answer. Diode pins is the proposed solution for MCU legs, but a better one IMO is to get some copper or brass wire with the correct diameter to match the low profile sockets. You get a spool for a few dollars and it is more convenient to use compared to getting diode pins with a constant diameter. It is still a bit of a pain to populate all holes of the MCU but not as much as cutting diode legs to proper length. With a spool you have more freedom and have plenty of spare. Hope this helps.

hoijui, 2025-08-14:
today I finally put together the laser-cut plywood case. there is still a lot wrong with it, but now I have an overview/an idea how to finish it.
I did end-up getting a copper wire, but.. it took a few attempts. according to the specs, the diameter of the MCU socket holes is 0.89mm. I tried 0.85mm copper wire first, but that would not go in at all. then I tried 0.8mm, which required a bit of force, but not much. the issue though, is that it is not impossible to remove the board from the socket, because... while it is possible to pull out a single wire from its whole, it is not possible to pull out the board with all its 24 PINs without cracking the board into pieces. so.. my next try will be with 0.7mm copper wire. if that will be too loose, I will maybe use 0.8mm for 4 holes, and 0.7mm for the rest.

hoijui, 2025-08-14:
... ouh, and yes, the socket + MCU sticks out by about 3mm in total from the bottom of the stacked case.

I have no special desire to have the keyboard very thin, so that is Ok for me, but I have to add an other layer to the case now.